ericrosenbaum/BeetleBlocks #81: Update the "about..." dialog

bromagosa, 2015-01-08:
It should give information about both Beetleblocks and Snap!

bromagosa, 2015-01-08:
Snap! 4.0
Build Your Own Blocks

--- beta ---

Copyright 2014 Jens Mönig and Brian Harvey
jens@moenig.org, bh@cs.berkeley.edu

Snap! is developed by the University of California, Berkeley with support from the National Science Foundation MioSoft, and the Communications Design Group at SAP Labs.

The design of Snap! is influenced and inspired by Scratch, from the Lifelong Kindergarten group at the MIT Media Lab for more information see http://snap.berkeley.edu and http://scratch.mit.edu

gitduks, 2015-01-19:
This is a first pass at what the 'about' window could say. Please take a look and edit it as you see fit.

Beetle Blocks 0.1 alpha

Copyright 2015 Duks Koschitz and Eric Rosenbaum
duks@pratt.edu, eric.rosenbaum@gmail.com

Beetle Blocks is developed by Duks Koschitz and Eric Rosenbaum with
support from Pratt Institute and the Media Lab at MIT. Jens Mönig and
Bernat Romagosa have been instrumental in realizing this release.

---

Beetle Blocks is based on Snap! 4.0 beta
Build Your Own Blocks
Copyright 2014 Jens Mönig and Brian Harvey

Snap! is developed by the University of California, Berkeley with
support from the National Science Foundation MioSoft, and the
Communications Design Group at SAP Labs.
The design of Snap! is influenced and inspired by Scratch, from the
Lifelong Kindergarten group at the MIT Media Lab for more information
see
http://snap.berkeley.edu and http://scratch.mit.edu

bromagosa, 2015-03-09:
Hey, I haven't done this yet because I was waiting for confirmation. Personally, I like the [text in the Github readme.md](https://github.com/ericrosenbaum/BeetleBlocks/blob/master/README.md) file better, starting at ``BeetleBlocks is a project by...`` but of course that's up to you :)

bromagosa, 2015-07-15:
We've decided to have an "About BeetleBlocks" dialog with an "About Snap!" button that opens the original Snap! dialog.