ericrosenbaum/BeetleBlocks #84: Rename blocks in 'Shapes' category

gitduks, 2015-01-16:
The general convention is:
The length is L:
The width is W:
The height is H:
Size is now dimension and is abbreviated to Dim.
Diameter is Dia.

<cube size (0.5)> should be <cube Dim. (0.5)>
<sphere diameter (0.5)> should be <sphere Dia. (0.5)>
<text [hello] height(1) depth(0.5)> should be <text [hello] H: (1) W:(0.5)>
<set extrusion diameter to (1)> should be <set extrusion Dia.to (1)>
<change extrusion diameter by (1)> should be <change extrusion Dia. by (1)>

bromagosa, 2015-01-16:
Hey Ducks, I think something went wrong in all of your recent comments, here's what I see:

[image]

gitduks, 2015-01-16:
Sorry see above...

commit 78e93fef (2015-01-19): Fixed #84
closed 2015-01-19 by bromagosa