ericrosenbaum/BeetleBlocks #85: Center the text of the text block

gitduks, 2015-01-16:
There are a few questions regarding the text. I am thinking of using the cuboid as a 'reference' for the rationale for placement and size.
Right now the text gets created with a lower left corner alignment and it might be better to compute the width of a string and then center it relative to the beetle.
The length could also be centered.
The text could be aligned in the inverse direction (as the beetle moves forward it writes 'left to right' while looking forward - right now it is writing backwards)

bromagosa, 2015-01-19:
Completely agree with this. Actually, the first time I used the text block I was expecting the exact behavior you describe (centered, left to right).

Should I go on and implement this?

gitduks, 2015-01-19:
Yes, that would be great!

bromagosa, 2015-01-20:
I mimicked the way cuboids and tubes are placed, but before I push the changes I want to be sure this is exactly what you want. Here's a couple of shots of how it works now:

[image]

[image]

closed 2015-01-20 by bromagosa

gitduks, 2015-01-20:
This is interesting. Didn't think of it that way. We could leave it like this or rotate the text 90 degrees around the z-axis as if you were reading it left to right and extruding along x.

bromagosa, 2015-01-21:
Completely up to you! Your way is one line of code shorter, if that helps your decision!

commit 3f97e8b1 (2015-01-20): fixed #85