ericrosenbaum/BeetleBlocks #88: Lines are not exported in OBJ files

bromagosa, 2015-02-02:


bromagosa, 2015-02-02:
The funny part is that vertices are exported, but lines in between these vertices do not show up:

[image]

[image]

bromagosa, 2015-02-10:
I'm reformulating this issue, as we've found out OBJ doesn't support things without volume.

bromagosa, 2015-03-10:
@gitduks these are the formats we can easily export into:

https://github.com/mrdoob/three.js/tree/master/examples/js/exporters

Could you check whether SceneExporter or TypedGeometryExporter are what we need?

gitduks, 2015-09-11:
I don't think these will work. Did someone look into any of these:
.dxf, .3ds, .3dm?

closed 2015-12-22 by bromagosa