ericrosenbaum/BeetleBlocks #89: Extruding

nemilya, 2015-02-10:
Hello!

I'm investigate this great 3d tool.

Can you give me some examples of Extruding in BeetleBlocks?

Is it possible to make like this "difference": https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/CSG_Modelling#difference

nemilya, 2015-02-10:
I've found some examples in "Open Project" - it's clarified.

Is Extruding is like "3D printer extruder" http://en.wikipedia.org/wiki/3D_printer_extruder ?

ericrosenbaum, 2015-02-11:
Thanks for trying it out! Glad the examples helped. Maybe "extrude" is not the clearest word. The Beetle can leave a 3D tube behind it as it moves... maybe it is more like a spider spinning silk or something.

Having CSG differences would be very cool! We don't have a way in Beetleblocks to refer back to a particular shape after it has been created, which you would need to select a shape and subtract it from another shape. But we could create a mode where any shapes you make are "negative," so if they overlap existing shapes they would have chunks subtracted out of them... I don't know how difficult the algorithm is for this, but it would be very cool and useful if somebody could implement it.

bromagosa, 2015-02-11:
Apparently it's possible: 

http://learningthreejs.com/blog/2011/12/10/constructive-solid-geometry-with-csg-js/

I'll check it out when I have some more time, I also think this would be really useful.

nemilya, 2015-02-11:
Yes, looks interesting: http://evanw.github.io/csg.js/

nemilya, 2015-02-14:
Also http://michaelmak.es/three-js-and-threecsg-js/ 

And https://github.com/chandlerprall/ThreeCSG

https://github.com/chandlerprall/ThreeCSG/blob/master/examples.html

Btw, what do you think about adding OpenScad block? Works on openjscad.

nemilya, 2015-02-14:
And this looks interesting
https://github.com/EiNSTeiN-/openscad-js

bromagosa, 2015-02-16:
Adding a new issue with the negative geometry feature request, for discussion.

closed 2015-02-16 by bromagosa