ericrosenbaum/BeetleBlocks #90: Trig errors?

ericrosenbaum, 2015-02-11:
This project demonstrates a very strange problem. Apparently combining a lot of 90 degree rotations can sometimes create large errors in the resulting angle... It seems to happen only if the rotations are aligned with the global axes. Maybe it's related to behavior of trig functions at 0 degrees? 

One stack in the project creates a cube fractal that shows the error (top cubes are tilted! see screenshot). Another stack with a tiny initial rotation (but otherwise identical) gives the expected behavior (cubes behaving themselves cubely). 

http://ericrosenbaum.github.io/BeetleBlocks/beetleblocks.html#present:Username=ericrosenbizzle&ProjectName=cube%20fractal

[image]

bromagosa, 2015-02-11:
I got this exact bug a while ago, but I thought it was my script that was wrong!

I'll take a look as soon as I can.

bromagosa, 2015-03-09:
Funnily, this works:
[image]
but this doesn't:
[image]
Apparently, three.js is rounding numbers internally.

bromagosa, 2015-04-01:
http://ericrosenbaum.github.io/BeetleBlocks/beetleblocks.html#present:Username=bromagosa&ProjectName=bug

So my project works in my home PC in Firefox, but not in Chrome.

Eric's project doesn't work anywhere in my home PC, but it does work at my work PC... :(

bromagosa, 2015-07-15:
Yay! Fixed by updating to the latest three.js version!

closed 2015-07-15 by bromagosa