ericrosenbaum/BeetleBlocks #91: Negative geometry

bromagosa, 2015-02-16:
There has been a request for a mode where any shape created is negative, meaning it creates holes in whatever other geometry it intersects with.

Is there any interest in this?

gitduks, 2015-02-17:
Am not sure about this. Eric and I should have a conversation about this I think. Generally I am in favor of adding features and functionality, but this might be complicated as we don't usually 'call' a geometric object. Having lists of geometric objects might be interesting, though...

bromagosa, 2015-04-01:
I've got a __very__ early prototype kind of working, but the resulting geometries are ugly as hell!

Check this out, it's just a cube from which I subtracted a sphere, then I placed a smaller sphere in the same position as the negative one:

[image]

Just look at the wireframe... it's a disaster:

[image]

This was achieved by using [this library](https://github.com/chandlerprall/ThreeCSG). I'll keep digging to see if there's something better out there...

bromagosa, 2015-04-01:
Forget it, it may work!

[image]

gitduks, 2015-04-01:
Very cool! If it works for cuboids and spheres we can already do a lot!!

nemilya, 2015-04-01:
Thanks! Looks interesting

ericrosenbaum, 2015-04-15:
how about 3 blocks:

subtract cube dim. []
subtract sphere diam. []
subtract cuboid l: [] w: [] h: []

ericrosenbaum, 2015-04-15:
Lets clip scale at zero, so it can never go negative

Remove the negative geometry blocks from the palette

Detect when you make a cube, cuboid or sphere with a negative dimension, and switch to negative geometry mode

Then we can add a block in a library that wraps this functionality - basically invert the dimensions you give it

bromagosa, 2015-04-23:
I'm closing this, as we do kind of have negative geometry already.

Let's open new issues for bugs and improvements to this.

closed 2015-04-23 by bromagosa