ericrosenbaum/BeetleBlocks #92: Loading a project via #present disables buttons under the Stage

bromagosa, 2015-02-16:


gitduks, 2015-02-28:
It seems that the 4 switches don't work in older projects. This project also does not show the variables in the area under the 3d window.

http://ericrosenbaum.github.io/BeetleBlocks/beetleblocks.html#present:Username=descomp&ProjectName=2D%20Hatch%20Square

bromagosa, 2015-03-02:
You mean this does work in newer projects?

gitduks, 2015-03-02:
It happens often when we use shared files in class. Some of them I made right before the class started in January. If you start a new project it seems to be fine.
The variables not showing up seems to occur on new projects that have been played with a lot, too.

closed 2015-03-09 by bromagosa

bromagosa, 2015-03-09:
This one was hard to track! Since it was a problem only showing up when loading from the cloud, I couldn't test it in local...

Anyways, it's been fixed now in c84f9da0c11570131b99705b33494ad025eb8fde.

commit c30d2721 (2015-03-09): trying to fix #92
commit b5ee84c8 (2015-03-09): still trying to fix #92
commit c84f9da0 (2015-03-09): and still trying to fix #92