ericrosenbaum/BeetleBlocks #101: Copy/paste stacks across windows?

ericrosenbaum, 2015-03-15:
We would like an easy way to copy and paste stacks of blocks across projects.

One way would be to use the OS clipboard.

You would click a stack, and in the context menu select a "copy stack to clipboard" item (it needs to be clearly different from "duplicate"). The OS clipboard would then contain a text string representation of the stack. There would also be a "paste stack from clipboard" when you click the scripting area that puts the blocks in that location.

Another way would be to create something like the Scratch "backpack." This would be a little pane below the Beetle info area (with its own refresh button). You could drag a stack into it. If you then switch to another Beetleblocks tab, you could then click refresh, and drag out the stack. I think it would be okay if it can only contain one stack at a time, and a new stack replaces the previous one.

There are some tricky aspects to copy and pasting across projects. The main one is that stacks can refer to variables and user blocks. So probably anything you copy should bring with it all of those. Blocks can include other user blocks, so they would have to be searched recursively for those references.

bromagosa, 2015-03-17:
I spent a long while trying to fix the copy-paste issues with Snap!, and I would say the OS clipboard approach is almost impossible... even huge sites like GitHub are forced to use a small piece of embedded Flash to achieve similar behaviors, because the browser forbids you to access the OS clipboard.

bromagosa, 2017-02-21:
This is what I came up with:

[image]

commit e0c82a87 (2017-02-21): fixes #101
closed 2017-02-21 by bromagosa