ericrosenbaum/BeetleBlocks #102: Remove green flag?

ericrosenbaum, 2015-03-15:
It's not clear that there's any real use for the green flag in beetleblocks. Should we remove it altogether? The green flag button above the stage would also go away.

bromagosa, 2015-03-17:
Wow are you sure?

Having starting points is essential for parallelization (okay, we can spawn threads in Snap! but this isn't obvious to beginners).

Also, a script without a hat block looks kind of ugly, don't you think?

jmoenig, 2015-03-17:
I'm with @bromagosa here. especially when presenting a project - or when sharing a project url - it's nice to automatically trigger the green flag, so you get to see that something actually happens. Also, quite often there isn't a *single* script that's running by BeetleBlocks project, but several parallel ones, and for this it's convenient to use the green flag.

closed 2015-04-12 by ericrosenbaum