ericrosenbaum/BeetleBlocks #95: Import 3D geometry file as shape primitive

ericrosenbaum, 2015-03-15:
We're imagining a way to import a file with geometry in it, such as a DXF, so that you can position copies of it in the world. So there would be a block that would place the imported geometry at the beetle's current location, like the existing shape primitives.

One design could be to have a block with a menu listing the imported shapes (starting with none), and an item called "import..." that brings you to a file dialog. (this is inspired by the "play sound" block in scratch, which has "record..." entry). It could look like:

imported shape [menu]

bromagosa, 2015-05-13:
These ideas came out:

* Have them work as Snap! costumes
* Import them as blocks
* Import them as dropdown menu options in an "imported shape" block

How to preview them, how to delete them, how to later export projects that have these shapes

bromagosa, 2015-12-03:
Aaaand following #197, when you import an OBJ file, the beetle wears that shape and can be stamped around.

gitduks, 2015-12-04:
Eric and I had similar conversations in the past, but thought of them slightly differently. The idea was to think of a 'state' and then use geometry primitives. So the beetle is in 'cube state' and then drops it for example. Another idea was to have the beetle be in a 'line state' which would be a line that goes left to right. Once the beetle moves it would 'extrude' that line as a ruled surface. We had also thought about the placement control of imported geometry the way you describe it.

We should definitely implement something like that and can maybe discuss the best visual metaphors for the naming and behavior of the blocks. I guess importing obj's is easy as you did that with the beetle itself. I am not sure if you could look into dxf file types again, but that would allow us to import 3d lines.