ericrosenbaum/BeetleBlocks #96: Zoom to extents

ericrosenbaum, 2015-03-15:
Set the camera to a zoom level that lets you see all the geometry in the scene (without changing the orientation of the camera). This probably should be a button near the stage eventually, but for the moment it can go in the camera menu.

bromagosa, 2015-04-08:
This is working now. If you feel the camera is way too far from the object, I can adjust it.

gitduks, 2015-04-08:
It also looks like the camera is keeping the same height. This means, that when it moves away it appears relatively low. You could also keep closer as you said.

ericrosenbaum, 2015-04-12:
Here's an idea: the zoom to fit operation only moves the camera toward or away from its target. the orientation of the camera is always preserved, as is the target (i.e. if you have panned away from the origin). it moves forward or back so that all geometry is inside the view with a small margin around the edges.

A slightly more complex version of this would start by calculting the centroid of all the geometry, setting the camera's target there, then moving back until the edges of the geometry's bounding box are inside the view.

bromagosa, 2015-04-29:
Keep rotation, use the bounding sphere.

closed 2015-06-17 by bromagosa