ericrosenbaum/BeetleBlocks #98: Add examples to file menu

ericrosenbaum, 2015-03-15:
I would like the examples to be much more obvious, so that people are more likely to find them as they get started.

One idea is to create a new file menu item that brings you to the same dialog as the "open..." item does, but with the examples shown.

When we have a landing page, that can also include links that directly open example files.

nemilya, 2015-03-15:
Yes, examples will be great. And adding links to shared projects to start page - will be good too.

My small experiment - 'Hi':
http://ericrosenbaum.github.io/BeetleBlocks/beetleblocks.html#present:Username=nemilya&ProjectName=beetleblocks-hi

Was done for 3d printing.

bromagosa, 2016-09-01:
Closing issues that are solved or rendered invalid by the BeetleCloud.

closed 2016-09-01 by bromagosa