ericrosenbaum/BeetleBlocks #106: buttons and user interface

gitduks, 2015-03-25:
I have been meaning to upload this in a while:

- We could have two buttons for camera related controls.
- Six check boxes to turn things on and off.
- The x,y,z coordinates separated.
- I added the three window presets (we could keep the shift-click version of the mini)
- I think the one pixel border of the individual frames could be white in all cases
- I added scratch-like zoom buttons for the scripting area
- The camera button could go away (and the remaining items could go into the gear)

[image]

gitduks, 2015-04-01:
Here is an updated version of UI changes:
- the camera and cloud icons are gone
- all grey backgrounds are the same (I think that should include the one behind the categories)
- the maximize 3d window icon is on the left in order to get the sequence right

[image]

bromagosa, 2015-04-01:
Nice!

All options under _Cloud_ and _Camera_ should go somewhere else, though. I'm guessing bg color and grid options should go under the _Config_ menu, and all cloud stuff under _File_, is that right?

bromagosa, 2015-04-01:
Add **x: y: z:** to rotations too.

gitduks, 2015-04-14:
Here are a few new suggestions:
- thinner lines for the enlarge stage to full screen button (these could also be an outline of the current version)
- add a triangle to the menu buttons on the left in order to distinguish them from regular buttons.
- add a white line between category selector and blocks panel.

[image]

bromagosa, 2015-04-15:
I'm slowly getting there...

closed 2015-04-30 by bromagosa