ericrosenbaum/BeetleBlocks #109: Remove save to browser

bromagosa, 2015-03-25:


nemilya, 2015-03-26:
Please explain - if some one clone and open beetleblock - where will be possible to save?

For example -http://nemilya.github.io/BeetleBlocks/default.html - now not possible to save to cloud, only local.

May be with Shift+click?

jmoenig, 2015-03-26:
good point, @nemilya . Maybe we should reconsider. Thanks!

cycomachead, 2015-03-26:
@jmoenig This issue applies to Snap<i>!</i> as well, I think.  If you come up with anything nice then I think it could be applied to both projects! :)

bromagosa, 2015-03-26:
Okay, just a thought here, and actually just a little hack that seems to work properly.

What about just changing the datatype of the URL? Something like ``data:snap``, for instance, won't make the browser think it has to intepret the data and, thus, will open a save to disk system dialog.

It works in local, should I issue a pull request?

jmoenig, 2015-03-26:
wow, I did not know that works.

bromagosa, 2015-03-27:
@jmoenig, should I issue a pull request with this change or you prefer to test it a bit more?

I've been trying it in local and found no problem, block libraries and projects are exported and imported properly.

I have to agree it's a little bit of a hack, but hey, it works! :)

jmoenig, 2015-03-27:
let's discuss this next week. I think @nemilya has a point about keeping it around. Also, it might be good to be a little reluctant before we storm ahead and react to "confused" users, as these voices might night reflect the majority...

bromagosa, 2015-03-27:
Oh wait, I think we're discussing different issues here!

My comment on changing ``data:text`` to ``data:snap`` was for the _Save to disk_ option. If you reuse exactly the same code you have under _Export project_ but switch from ``data:text`` to ``data:snap``, you'll get _Save to disk_ to work independently of project size and, apparently, browser/system.

Sorry, I guess this discussion should be taking place in #749 or #726 in Snap's main branch.

jmoenig, 2015-03-27:
wait, what? So, this replaces the "save to disk" gimmick, is that what you're saying? Well, if if even addresses the crashing problem (how?) let's give it a try, by all means!

bromagosa, 2015-03-29:
Yep, that's what I'm saying! I don't know if it addresses the crashing problem, but I'll try tomorrow and let you know :)

bromagosa, 2015-07-15:
@jmoenig @ericrosenbaum @gitduks what should we do with this one?

bromagosa, 2015-07-15:
Let's rename save to disk to export to disk

commit 192d5ad3 (2015-09-14): fixed #109
commit 5db27a07 (2015-09-14): fixed #109
closed 2015-09-14 by bromagosa