ericrosenbaum/BeetleBlocks #111: change background color to white?

jmoenig, 2015-03-26:
is there a reason why the 3D stage's background color is gray? Personally I would prefer it to be white per default

cycomachead, 2015-03-26:
I've only toyed around with BeetleBlocks, and idk if there is an exact reason - but how important is color accuracy for users? (since I never intend to actually print anything for the time being, it doesn't matter to me! :P) However, generally grey is a better background color if you are trying to get correct looking colors for a print of some sort.

gitduks, 2015-03-26:
The reason is just based on CAD standards or conventions. 3d software used to have black or white backgrounds and then switched to gray as it appears to be less straining.
Could we maybe include background color settings in the grid options? I am sure users will have their own preferences in general. Personally I would like the default to be a gray that we use some other place in the UI.

jmoenig, 2015-03-27:
It's not a big thing at all since we already *can* set the background color in the "camera" menu, it's just that the gray background somehow feels strange to me, given that the rest of the UI tries to be bright (at least the background colors), and that the gray background doesn't seem to carry any additional information. I've also noticed that the RPI folks (Ron Eglash etc.) who are also using three.js have also switched to white. But that's purely anecdotal, of course...

bromagosa, 2015-03-27:
What about having bg color preferences persist like language and block size do? That shouldn't be too hard to implement.

gitduks, 2015-03-30:
That would be great, actually. Shall we turn this into a feature?

closed 2015-07-14 by bromagosa