ericrosenbaum/BeetleBlocks #112: add pinch-zoom to 3D stage

jmoenig, 2015-03-27:
so BeetleBlocks becomes fully usable on iPad and tablets. I think I need to enhance Morphic.js for this. Investigating...

bromagosa, 2015-03-27:
Yes, this is very much needed!

Also, the tap behavior is not working quite well. I think (please correct me) morphic.js expects the mousedown and mouseup events to occur at exactly the same pixel, otherwise it understands a drag is happening. Maybe there should be a small threshold to account for not pixel-perfect fingers ;)

jmoenig, 2015-03-27:
I've already looked into the issue of adding two-finger movement for zooming and three-finger movement for panning, and it looks like it's pretty straightforward to implement.

I'm somewhat reluctant to introduce a threshold, because while that's going to improve tapping it is also going to make dragging less responsive. But I do have a few other ideas how we can have our cake and eat it, too...