ericrosenbaum/BeetleBlocks #116: Add URL parameters to allow embedding

bromagosa, 2015-04-01:
https://github.com/jmoenig/Snap--Build-Your-Own-Blocks/pull/744

bromagosa, 2015-04-08:
This is now working! To embed a project, follow these steps:

1. **Get the shared URL**
2. **Add the desired parameters**
    * _hideControls=true_: do not show any controls and remove the "Are you sure you want to leave?" dialog (this is what you usually want when embedding)
    * _noRun=true_: do not fire the green flag event
    * _editMode=true_: open the project in the editor

3. **Wrap it in an iframe with the desired dimensions and parameters**

So, for instance, the final HTML markup for a project of mine would be:

    <iframe src="http://ericrosenbaum.github.io/BeetleBlocks/beetleblocks.html#present:Username=bromagosa&ProjectName=gear&hideControls=true" width=640 height=480 frameBorder=0></iframe>

Now see it in action: (example page to be removed later)
http://ericrosenbaum.github.io/BeetleBlocks/homepage/embed.html

closed 2015-04-08 by bromagosa

gitduks, 2015-04-08:
This is so cool!