ericrosenbaum/BeetleBlocks #122: Color scheme and Blocks Category Selector

gitduks, 2015-04-12:
We have looked at a color scheme again and Eric pushed this version. Bernat, please let us know if you want the regular color scheme for your workshop!

Here is some of the rationale behind this darker color scheme:
Orange tones remain for variables and lists.
Green tones remain for operators.
Blue tones apply to motion, shapes and 'my Blocks'
Dark grey is used for control.
Zebra does not invert text to black.

I would like to use this in my seminar to get feedback form my students about this. It would be great to change the text color of the selected category in the Blocks Category Selector

bromagosa, 2015-06-10:
There is an issue with the current color scheme when you embed control blocks inside a launch block:

[image]

gitduks, 2015-06-20:
I guess we need to look at all blocks that are grey. I think we could make the default grey color a little lighter.

closed 2019-11-09 by bromagosa