ericrosenbaum/BeetleBlocks #125: boolean reporter <inside?> to check if beetle is inside of geometry?

ericrosenbaum, 2015-04-13:
Right now the beetle is kind of "blind," since it can't detect or interact with any of the geometry it has created. One simple way we could start doing this is to have a boolean reporter (in the sensing category) that returns true if the beetle's current coordinate point is inside of any of the geometry. 

I'm not sure if this is computationally easy or hard in general. I can see how it's easy with spheres (just check the distance to the center point of each sphere, and check if that is less than the sphere's radius). cubes and cuboids are probably do-able, though if they are rotated off axis that gets tricky I guess. but maybe there's a general solution to check if a point is inside a mesh? 

Partial solutions that would still be super useful would be:

<inside any sphere?>

<inside any sphere, cube or cuboid?>

bromagosa, 2015-04-14:
When implementing negative geometry I looked for a way to detect intersections so we could compute interactions only between touching objects, and there aren't many examples out there, so I'm guessing it is not trivial. Three.js does have a raycaster that could help here.

bromagosa, 2015-09-09:
I've been taking a look at this, it seems like a really HARD problem:

http://stackoverflow.com/questions/16361327/check-if-point-is-inside-a-custom-mesh-geometry

In our case, using the bounding boxes or bounding spheres is not enough.