ericrosenbaum/BeetleBlocks #135: Can we export STL files with arbitrary resolution?

jgrizou, 2015-04-23:
We are currently develloping a library to build simple some shapes in OpenScad ([robotis-scad](https://github.com/jgrizou/robotis-scad) used to build [poppy-ergo-jr](https://github.com/poppy-project/poppy-ergo-jr)). 

We target education purposes and would like kids to play around with such parts to create their own robots. OpenScad is too complex for kids, BeetleBlocks seems to be the perfect alternative. 

Yet it would be nice if we could choose the resolution of the export. Some parts, especially curved ones or parts with holes, need more resolution to be usable once 3D printed.

In OpenScad, we can change the %fn variable to generate more or less detailled STL files. Is there a way to do something similar?

bromagosa, 2015-04-29:
@jgrizou we'll get back to you soon with news about that :)

ericrosenbaum, 2015-04-30:
added feature request #136

bromagosa, 2015-10-28:
Duplicate (#136)

closed 2015-10-28 by bromagosa