ericrosenbaum/BeetleBlocks #137: A way to "talk" to 3D objects

bromagosa, 2015-04-30:
It may not meet Beetleblock's original purpose, but it would be great for animations and other applications.

Imagine being able to do something like this (we should really think about a proper syntax here):

[image]

Anyway, this is something to discuss, the syntax is not trivial, and I'm not sure you guys want to have something like this in the system either.

ericrosenbaum, 2015-04-30:
wow! hmmmmm. I guess the [object {sphere}] construction both runs the sphere block (putting a sphere mesh into the world) and returns a kind of "pointer" to that particular mesh. maybe it should be called "mesh object," even though that's a bit obscure. 

I guess we would need a beetleblocks class that wraps the three.js internal mesh class, so we could both set and get the mesh's position, rotation and color.  

One weird thing is- what would a mesh object look like in a monitor? maybe it can just have a string representation that shows the properties.

I suppose it would be possible to access the list of vertices of the mesh too? maybe even modify them??

jmoenig, 2015-05-01:
Okay, this really calls for an object model that's somewhat similar to what sprites are to Scratch, except that Beetleblocks creates the programmatically. Thinking about it some more this could totally drop out of my current work on the new OOP system for Snap, where we'll get first-class sprites and slot-wise inheritance (prototypes). This is exciting to think about!

gitduks, 2015-05-01:
wow, very interesting -  I may not exactly understand the object in this case. How would we have access to the mesh object? Is the pointer something that a user could see, rename or put into a list?