ericrosenbaum/BeetleBlocks #138: SVG export

ericrosenbaum, 2015-04-30:
I'd like to include SVG export as a feature of beetleblocks. It would go in the file menu, with the other file export options.

To do this, we would just need to keep track internally of all lines that have been drawn. This could be as simple as appending to an array of xy coordinate pairs (discarding z axis information), each time a line is drawn. ideally we would store the color of the line as well. the array of course would be cleared on reset. Then the export function would assemble the SVG string.

The simplest implementation would create a new SVG line for each line drawn in beetleblocks. an improvement on this would be to assemble a polyline when the lines are connected. in general, if drawing is on, any time you move the beetle you make a line, so the result should be one connected line, which could be made into a single polyline. there is an exception though - push and pop move the beetle without drawing a line. I think this behavior makes sense, but it means we can't automatically always have a single polyline for each draw on / draw off. I'm not sure what the best solution is- maybe the list of points stored internally could have markers in it for the start and end of each line? 

actually - another solution would be to actually use the internal threejs representation of the line geometry to generate the SVG. that might actually be better and simpler!

ericrosenbaum, 2015-05-03:
working on this now!

ericrosenbaum, 2015-05-03:
okay, I added a first working version of it. It uses the internal threejs geometry to generate an SVG line for each line, including its color. I discard the z coordinates, and scale up by 10x.

the result is rotated and flipped, but that should be easy to fix.

after that, the next step is to use polylines where possible. one way to do this would be to check if the second point of a line is identical to the first point of the next line, and if they are, start or continue a polyline.

bromagosa, 2015-05-03:
Very nice!

I noticed it only exports lines on the positive quadrant though, is this the intended behavior?

gitduks, 2015-05-03:
Wow cool!
If you scale it by 72 I think we get 1 BB unit = 1 inch.

ericrosenbaum, 2015-05-03:
okay I fixed the rotation, and set the scale as @gitduks suggest to 1BB unit = 72px = 1in

@bromagosa I'm not sure what you mean, sorry! ideally it would export all lines. can you send an example?

bromagosa, 2015-05-08:
It seems it's just a browser issue. When you open an SVG file in a browser, only the positive side (x>0, y>0) is displayed. The whole shape is there however!

About the conversion to inches, I think we should provide with some kind of export dialog, or preferences dialog, so people can choose their own units. This goes for STL and OBJ exports as well.

closed 2015-05-13 by ericrosenbaum