ericrosenbaum/BeetleBlocks #139: Same instructions that draw a square extrude a pentagon

envelupo, 2015-05-09:
In the capture attached you can see both the pentagon and parts of the square. 

Using Firefox 37.0.2, w7-64

[image]

bromagosa, 2015-05-12:
Indeed, this is a nasty bug!

Thanks for reporting it.

bromagosa, 2015-05-12:
Just so you can work on your project in the meantime, extruding single lines does output the expected result:

[image]

envelupo, 2015-05-12:
Thanks for the workaround :)

bromagosa, 2015-06-03:
Ok, I've been investigating for hours and found out the problem is not with the ``SplineCurve3`` curve, I've rewritten the extrusion by turning it into an actual circle shape extruded along the path and I got exactly the same issues with sharp angles.

There are several possible solutions, please pick your favourite!

### Circle extrusion with 50 segments per curve:

[image]
Pros: Sharp angles look better
Cons: Looks faceted. Doesn't actually look like a square.

### SplineCurve3 with lots of segments in its tube geometry
[image]
Pros: Sharp angles look better. Doesn't look faceted.
Cons: Doesn't actually look like a square.

### ClosedSplineCurve3 with reasonable amount of segments in its tube geometry
[image]
Pros: All former problems solved
Cons: It's a closed shape! We'd need a different kind of extrusion mode for this.

There is another possible solution, which is to detect when a shape has been closed (end point equals start point with a threshold) and turn the Spline3 into a ClosedSpline3. That, though, would mean we'd need to think about what to do with future points added to the extrusion path...

bromagosa, 2015-06-03:
Let's do tube → sphere → tube

bromagosa, 2015-06-09:
This has been fixed, and all examples tested to work with no problem except for when there are A LOT of extrusion points, like in the snail example, which now looks a bit uglier and makes BeetleBlocks quite slower in my PC:

[image]

Still, everything should now be printable, which I think was the priority.

BTW, I've used cylinder geometries because they're native to three.js and look smoother than extrusions.

Let's now think of ways of speeding this up! :)

commit 826c29e5 (2015-06-09): fixes #139, for now!
closed 2015-06-09 by bromagosa