ericrosenbaum/BeetleBlocks #145: embed code for thumbnail image link

ericrosenbaum, 2015-05-13:
the one-click share system also gives you an embed code that displays the project thumbnail that links to the shared project

bromagosa, 2015-06-09:
@jmoenig any news on the API call to retrieve a project's thumbnail?

jmoenig, 2015-06-09:
working on it!

(also working on opening the Snap Cloud for other applications, such as Snapi)

-Jens

P.S. Great news about hardcoding codifications into Snap4Arduino today!

bromagosa, 2015-06-09:
Oh wow! I can't tell you how happy people in my team will be to hear that! Thank you!

bromagosa, 2016-02-15:
I think all these should be moved to the separate sharing site project. For now, https://github.com/bromagosa/beetlecloud could do.

closed 2016-02-15 by bromagosa

cycomachead, 2016-02-15:
@bromagosa You might find this useful: https://github-issue-mover.appspot.com

It closes the original issue and tags it, so everything is still linked.

bromagosa, 2016-02-16:
Yep, thanks! I recently used it to move all Snap4Arduino issues into a new repo.

In this case, I guess I'll just start anew :)