ericrosenbaum/BeetleBlocks #147: tutorial pane

ericrosenbaum, 2015-05-29:
We need something inside of beetleblocks to introduce new users to its functionality and guide them through some simple examples. I like the model used by Scratch 2.0, which shows a simple step-by-step tutorial inside a little pane the appears within the scripting area. It has a sequence of pages, with lots of animated gifs: 

[image]

I've begun drafting the text for these tutorials here:
https://docs.google.com/document/d/1M14XZqhr7gJOagaYyK445-DTSq2F_8nK9LDyKoNCzQ8/edit?usp=sharing

How would we implement this? It needs to display a set of pages, with text, images, animated gifs, links forward and backward, and a menu. Can we display HTML pages within the editor? 

There's also the question of where the pane goes. Maybe we could use the lower part of the scripting area, and it when closed it could fold down to a little bar at the bottom. It would open by default in many cases (e.g. not signed in, new user, open to blank project).

bromagosa, 2015-06-01:
See http://snapi.citilab.eu/#tutorial for my take on a Snap! tutorial. I just need a set of texts and PNGs to build a tutorial lite that for BeetleBlocks!

Oh well, that and some free time... :(

nemilya, 2015-06-02:
http://snapi.citilab.eu/#tutorial is great! Is it github'ed?

bromagosa, 2015-06-02:
@nemilya of course! The project is here: https://github.com/bromagosa/Snapi/
and the tutorial part is here: https://github.com/bromagosa/Snapi/blob/master/snapi/tutorial.js

Notice I've hardcoded the images in there, it doesn't have to be like this, I just needed to have this working ASAP and copy-pasted the base64 string straight away.

bromagosa, 2015-06-02:
Forgot to say, it only works in flat design mode!

bromagosa, 2015-06-03:
The tutorial should show up whenever BeetleBlocks detects it's never been opened in this browser before.

bromagosa, 2015-06-03:
What about animated GIFs?

closed 2015-10-28 by bromagosa