ericrosenbaum/BeetleBlocks #150: Grid issues

awaywiththespoons, 2015-06-10:
(working in firefox) 

When clicking inside the printing bed screen there is a problem with the grid -it redraws with every click until the whole screen is just "grid". I tried this in all the view modes and same thing happened.

[image]

bromagosa, 2015-06-10:
Wow! How did that happen? O_O

Can you point me to the steps you followed? I can't reproduce that in any of my computers or browsers...

awaywiththespoons, 2015-06-10:
hmmm. just had another go and it's not on click, it's mouse dragged that makes the grid go awol!

gitduks, 2015-06-10:
We have been encouraging the use of Chrome as we have not really tested any other browsers yet...

awaywiththespoons, 2015-06-11:
ahhh. I see. Just tested and it works fine in Chrome! 

I couldn't get anything to work at all in Firefox, maybe it would be worth noting that it should be used on Chrome in your warning in the try it now section?

bromagosa, 2015-06-11:
But wait, I actually use Firefox and Chrome alternatively all the time, and never had an issue... what OS, Firefox version, etc are you using?

awaywiththespoons, 2015-06-11:
oh that could be it. I'm not on yosemite. Firefox was updated a few days ago not sure of the verison.

bromagosa, 2015-07-15:
@awaywiththespoons is this still happening? We never managed to reproduce this bug...

bromagosa, 2015-12-03:
Since there's been no answer from @awaywiththespoons and nobody else complained about this, I'm closing this issue.

closed 2015-12-03 by bromagosa