ericrosenbaum/BeetleBlocks #154: Try out ExtrudeGeometry instead of cylinders + spheres

bromagosa, 2015-06-17:


ericrosenbaum, 2015-06-17:
Try rolling back to the spline extrusion technique, then adding additional points along each straight segment just before and just after the joint. hopefully this will make most of the path a straight line, and the region where it's curving is very small, and the resulting shape has less error.

ericrosenbaum, 2015-06-17:
Also, here's another idea: use use extruded tubes, but stitch them together at the joints. Each segment of the path is a straight tube. Each joint consists of the two circular ends of the tubes, which share an origin but may be rotated relative to each other in all three axes. Our current strategy is to put a sphere at this location, connecting the tubes because both circles lie on its surface. 

What if instead we connected each point one on circle to the nearest point on the other circle? We would then have a set of vertices which we could make into faces. There would be a discontinuity (a little curved elbow), but the whole thing could be a single solid object with many fewer extra vertices.

It would look something like this:

[image]

[image]
  
Here's my test:

http://ericrosenbaum.github.io/BeetleBlocks/beetleblocks.html#present:Username=ericrosenbizzle&ProjectName=circle%20join%20test

Note that this just connects the pairs of points on each circle "in order," so the stitching gets twisted when the joint has rotation in more than one axis. So instead, the algorithm should always find the nearest point on the other circle to connect to.

bromagosa, 2015-06-18:
Eric, nice example! Looks doable indeed.

Note that the correct URL is now: http://ericrosenbaum.github.io/BeetleBlocks/run/#present:Username=ericrosenbizzle&ProjectName=circle%20join%20test

gitduks, 2015-06-19:
Wow, this looks pretty clean. Any kind of mesh optimizer for 3d-printing should be able to deal with the vertices and edges inside the cylinders. Hope we can test this somehow.

commit 83a68d42 (2015-07-15): finally fixed #154
closed 2015-07-15 by bromagosa