ericrosenbaum/BeetleBlocks #157: Remove items from gear menu

bromagosa, 2015-07-15:


gitduks, 2015-09-28:
Remove:
Turbo mode
Prefer smooth animations
Flat line ends
Codification support

ericrosenbaum, 2015-11-18:
looking at the gear menu items, it seems like we can remove almost all of them. These are the only four that are really relevant:

Language...
Set background color
Set grid color
Set grid interval

it's ambiguous if the last three are user-level or project-level settings. Thinking about it I'd say they should really be project-level settings. I'll add another issue about that.

jmoenig, 2015-11-18:
I would keep Codification support. Once you find out what it does it's extremely useful for dramatically extending Beetleblocks in Beetleblocks itself. You can use it, for example, to compile scripts inline or to interface Beetleblocks to cloud data, robots, other 3D environments etc.

ericrosenbaum, 2015-11-18:
interesting... I thought the javascript block was what we would use mainly to extend? 

how do people find out what it does? enabling it has no visible effect.

jmoenig, 2015-11-18:
enabling Codification support adds some blocks to the Variables category and two entries to the context menu of template blocks sitting in the palette. the JavaScript function block is good for just calling JS, the Codification feature lets you compile scripts and translate them to any textual programming language. You can then either export the generated text code or even send it via the http block to something else to execute it. You can even compile scripts to JavaScript and then use the JS Function block to run them, that's how some people at CDG are interfacing to D3. We should probably discuss this in person, but Bernat himself has a lot of experience with this feature, he's using it to create embedded software for Arduinos and the new WhiteCat project.

cycomachead, 2015-11-19:
> You can even compile scripts to JavaScript and then use the JS Function block to run them, that's how some people at CDG are interfacing to D3.

Ok, I want to see this! That's pretty cool!

bromagosa, 2015-11-19:
Codification is very useful for advanced users... I'd keep it in there even if it's as a shift-click hidden option.

@jmoenig, for the WhiteCat project I'm not using codification, I don't even have threads anymore. Since the whole environment was meant to run in the hardware, it made no sense to keep any of those in the client, so I've stripped down Snap! to the bare minimum (blocks, ui, widgets...), then reimplemented things myself: https://github.com/bromagosa/WhiteCat/blob/master/dev/wc.js

jmoenig, 2015-11-19:
Oh, right, you don't even need an evaluator for WhiteCat, just an editor. Actually, this raises an interesting question, because I'm considering allowing certain things in block definition headers to be dynamic, such as drop-downs. That way you could specify the contents of a drop-down by providing a reporter or a script that computes it dynamically, which in turn requires access to the evaluator, if not necessarily the thread system.

bromagosa, 2015-11-19:
That would be great! This way the ``[ go to x: ( ) y: ( ) ]`` block could update its inputs automatically in the same way Scratch does, is that right?

(we've hijacked this thread, let's maybe move this discussion somewhere else?)

jmoenig, 2015-11-19:
Yes! and yes. :-)

bromagosa, 2016-06-02:
I kept what I believe is still useful!

commit 9da44e8a (2016-06-02): fixes #157
closed 2016-06-02 by bromagosa