ericrosenbaum/BeetleBlocks #158: extrude → move → rotate x → move → weird twisting

bromagosa, 2015-07-15:


ericrosenbaum, 2015-07-16:
It's not just x axis rotation that causes the twisting at the extrusion joints- it can also be a combination of y and z rotations.

bromagosa, 2015-07-16:
Is this still happening? I feel the fix I pushed today for extrusions should also take care of this...

ericrosenbaum, 2015-07-20:
Yup, this is still a problem:

[image]

bromagosa, 2015-07-28:
Check this one out, I think I may have mistaken an axis for another or something...

http://beetleblocks.com/run/#present:Username=ericrosenbizzle&ProjectName=code%20as%20form%20examples

bromagosa, 2015-09-14:
@gitduks had an idea. We can use the intersection between both circles as a start point, and then move (counter?)clockwise from there and stitch the corresponding points.

bromagosa, 2015-09-22:
After spending 10+ hours trying to fix the circle stitching issue (not kidding, I even dreamt about vertices tonight) I decided to look for alternative solutions and found out the latest three.js allows you to define the tension coefficient for Catmull-Rom curves.

What that means is we can achieve a pretty similar result for sharp edges with about 1/5th of the code, while smooth edges work much better now.

I've only found one example that wouldn't work well, all the rest look quite good.

[image]

I've adjusted it so the amount of segments is proportional to the length of the extrusion, so corners will look the same regardless of how long the path is:

[image]

An hexagon:
[image]

And our classical, smoother-looking examples:
[image]
[image]
[image]

It's not all a bed of roses though, the snail example is broken, but I can redo it to have it be a little bit less extrusion-abusing.

I believe this is the best extrusion we've had so far, and is a good compromise between sharp and smooth angles...

bromagosa, 2015-09-22:
Just a tiny change is required for the snail to look good again:
[image]

ericrosenbaum, 2015-09-22:
this is amazing! thanks for all your hard work on this. 

what exactly did you have to change in the snail example? is this something users will have to know in general to avoid a problem, or is it something we can prevent?

bromagosa, 2015-09-22:
Actually, I think I found the source of the problem!

gitduks, 2015-09-22:
Is there a way to make sure 90 degree turns are consistent? We should fix the upper right corner in this example, no? It also looks like the inside corner renders correctly at the bottom left and upper right, but not for the other corners.

[image]

bromagosa, 2015-09-23:
There should be a way, I guess. I need to investigate further.

For the sake of my sanity, I'll close this issue and open a new one about this particular problem.

closed 2015-09-23 by bromagosa