ericrosenbaum/BeetleBlocks #159: comments in script pic

gitduks, 2015-07-20:
It would be cool if the comments could be included in the script pic.

bromagosa, 2015-09-09:
[image]

You need to right click on the scripting canvas and select "scripts pic...".

Thing is, this exports _all_ scripts in the canvas, including comments. If you want to export a single script with its comments, you should remove all other scripts from the canvas before. I don't think there is a better way at the moment, but maybe @jmoenig will prove me wrong :)

bromagosa, 2015-10-28:
This was recently fixed in Snap!, so we got it fixed for free when merging.

closed 2015-10-28 by bromagosa