ericrosenbaum/BeetleBlocks #173: splines

gitduks, 2015-09-11:
I would be cool to have splines as an option to draw with. The type of spline could be a cardinal curve, which means that every beetle move results in a new a point on the curve:
[image]

The 'start drawing' block could include a drop-down menu where one can choose between 'start (line)' or 'start (spline)'.

bromagosa, 2015-09-14:
If we do this, it would probably make sense to have different extrusion methods as well.

gitduks, 2015-09-15:
Please take a quick look into this. You are right; there are several things to consider: have them show up on the screen, svg export and extrusion type. If this is hard, we can focus on other things.

gitduks, 2015-09-21:
Wow, that was fast!!! Have two questions for you. One is about the 'resolution' of the spline in terms of how it is rendered and the second is bout the orientation of the SVG:

Is there a way to set the preview of the spline to be smooth? Also, the spline is not going through the points the beetle is going to. If the 'resolution' is based on numbers, switching from odd to even (or inverse) might maybe work?

[image]

In the exported SVG the spline has the right behavior (see below), but it would be cool if we could rotate the drawing and set the artboard or canvas dimensions to 'inches' and not 'points'.

[image]

bromagosa, 2015-09-22:
The first one is just a matter of adding more control points to the splines in three.js. I'm doing that right now.

I think the second one should be discussed, because exporting in inches by default doesn't make much sense for the rest of the World. We either export to something agnostic (like pixels) or add an export dialog letting you choose units.

gitduks, 2015-09-23:
Yes, that is a good point! It would be very cool to have a choice between 'inches' and 'cm' in a dialog box for example. The rotation still applies to both cases though.

We also need to look at the location of the x=0 and y=0 of the canvas (see below). If you draw a square with lines, the canvas is exactly the size of the square and has the top left as its 0,0. In the case below the canvas is misaligned:

[image]

[image]

gitduks, 2015-09-28:
We could use the same 0,0 in BB as the 0,0 for the svg file. So in the above example the canvas would have it's origin where the beetle started to draw.

commit 0ecd2954 (2015-09-21): splines working, fixed #173
commit da536d9a (2015-09-22): increased spline resolution by 4x (fixes #173)
commit 30e23b96 (2015-12-03): proper exported SVG rotation and positioning (fixes #173)
closed 2015-12-03 by bromagosa