ericrosenbaum/BeetleBlocks #175: all saved projects with "start drawing" are broken

ericrosenbaum, 2015-09-22:
The change to the "start drawing" block that adds the menu for lines vs splines causes all projects saved with a "start drawing" block to break: the block loads with no menu selection, and the beetle does not draw.

Please remember to test!

ericrosenbaum, 2015-09-22:
I think the problem is in c92b0f2053c85d1c85bc795fe62414512da3ff6c

bromagosa, 2015-09-22:
Actually, they're not broken, they just show up with no option selected:

[image]

Which makes sense, as the old spec was ``start drawing`` and the new one is ``start drawing %drawStyle``, and ``drawStyle`` is ``null`` in the old blocks.

Of course all old projects should now be fixed and resaved, unless we want to add a particular case to the code that keeps backwards compatibility by doing that by default (not sure that's a good idea...).

ericrosenbaum, 2015-09-22:
I guess the real problem is that we have users already who are not us :/

if you look for example at the blog posts from the tinkering studio you can see that they have created dozens, maybe hundreds, of projects containing the "start drawing" block:

http://tinkering.exploratorium.edu/2015/08/25/beetleblocks-floor

now think about what happens to them: they open up a project, press the green flag, the beetle moves, and no lines are drawn. I would describe that behavior as "broken." 

it's true, for us, it's not such a big deal since we know what happened and why. But my strong preference is to prevent this type of problem for users if we possibly can. In this case, it's a pretty simple fix, isn't it? it's not backward compatibility that we need, exactly. we should just check for null, and apply a default if necessary (I don't see why all menus don't do that in snap blocks- but maybe there is a reason? I can't understand what the use case is for choosing "null" in a menu, especially when there's normally no way to do that without a bug. cc @jmoenig ).

bromagosa, 2015-09-22:
That makes sense indeed. Sorry, I hadn't thought about that!

I'll implement it so that if there is no option selected it still draws lines.

commit f5ad5c31 (2015-09-23): fixes #175
closed 2015-09-23 by bromagosa