ericrosenbaum/BeetleBlocks #179: Lines and extrusions do not respond correctly to color changes

ericrosenbaum, 2015-10-13:
When you start drawing with color 1, move, change to color 2, and move again, the expected resulted in that the line is drawn in two segments, the first in color 1, the second in color 2. Generally, you should be able to change the color of the line as you draw.

The current behavior is that color 2 is applied to the whole line. This is due to our recent change to line drawing behavior, combining line segments into a single polyline that can only have a single color. 

Extrusions have essentially the same bug (it has existed for a while).

I propose that we make a special case for color changes: 

when changing the color, check if we are drawing. if we are, stop and restart the drawing, so that a new line segment will be started, before changing the color. 

Similarly, we could also check if we are extruding, and stop and restart the extrusion so that the new extrusion will get the new color.

bromagosa, 2015-10-13:
That makes sense, but we'll have to make sure that splines and smooth extrusions behave the same way whether you add extra steps or not, which may not be the case!

ericrosenbaum, 2015-10-14:
It's true, splines and extrusions will not be continuous across the color changes. Instead, you'll get a corner, like between straight line segments. you can see this now by simulating the behavior I'm talking about, by adding stop and start drawing blocks at each color change. I don't think this is a big problem- the ability to change colors seems more important to me.

bromagosa, 2015-10-15:
Fixed for lines:
[image]
Extrusions will come after fixing #177

commit 42d1fdc8 (2015-10-15): fixed #179
closed 2015-10-15 by bromagosa