ericrosenbaum/BeetleBlocks #183: Tiles to allow creation of ball and socket joints

MrSteve2, 2015-11-05:
Would be fun if there was a tile to allow the creation of a socket to match a sphere so you could create ball and socket joints. Then kids could create something like a jointed puppy dog ex: http://www.thingiverse.com/thing:767995/#files

One way to do this would be to have a method to turn tiles into "cutters" instead of "drawers"
Then I could simply "cut" a sphere into a shape which would create a socket.  Also while computational geometry is wonderful, a man with a hammer can only nail.  Being able to cut or erase using other tools would allow a different way of creating objects that might be harder if done only with scripting.

In any case, great tool and lots of fun!!!

Cheers,
Mr. Steve

bromagosa, 2015-11-05:
Hi!

There is a hidden feature we don't encourage using because it's really (and I mean **really**) slow and experimental, but you can try it out and see if it fits your needs:

http://beetleblocks.com/run/#present:Username=bromagosa&ProjectName=ball%20joint&editMode=true

Basically, whenever you input a negative magnitude for some shape, BeetleBlocks will try to carve out that shape from whatever it finds around it.

I hope we'll eventually manage to make it smoother and faster...

closed 2015-11-05 by bromagosa

MrSteve2, 2015-11-05:
Thanks, works great.