ericrosenbaum/BeetleBlocks #185: Go To Block Should Break Before Axis Labels

maryfries, 2015-11-06:
The "go to" block should be breaking so that the "y:" and the "z:" are in front of their associated input slots—not on the previous lines.

[image]

maryfries, 2015-11-06:
See also: https://github.com/jmoenig/Snap--Build-Your-Own-Blocks/issues/999

jmoenig, 2015-11-06:
breaks within blocks are strictly automatic. Some languages don't prefix parameters in a keyword-like fashion, but instead suffix them. There is some internal support for user-defined breaks, which isn't yet exposed in the GUI. I'm thinking about it...

bromagosa, 2016-02-15:
I'm closing this, as Snap! already implements user-defined breaks, as shown [here].(https://github.com/jmoenig/Snap--Build-Your-Own-Blocks/issues/999#issuecomment-158833594)

closed 2016-02-15 by bromagosa