ericrosenbaum/BeetleBlocks #191: Input sliders should not show up in text boxes that are not block inputs

bromagosa, 2015-11-30:


jmoenig, 2015-11-30:
so, this is actually intentional, as a way to use Snap on touch devices

bromagosa, 2015-11-30:
oh wait, but does it make it possible to write text in there?

cycomachead, 2015-11-30:
> so, this is actually intentional, as a way to use Snap on touch devices

When the keyboard works, I don't know if it's necessary, but we could easily detect if someone is on a touchscreen and only show sliders then.

bromagosa, 2015-12-03:
[image]

Are we talking about the same thing? I don't understand how this affects writing text in touch devices.

closed 2015-12-03 by bromagosa

cycomachead, 2015-12-03:
I had thought the very old idea, was that even when you couldn't write text, you could at least write _numeric_ text. (Though I agree in general, the slider isn't necessary there.)

commit 8a39d447 (2015-12-03): fixes #191