ericrosenbaum/BeetleBlocks #192: New main menu design

ShuraChechel, 2015-12-01:
[image]

bromagosa, 2015-12-02:
As a general UI rule, shouldn't we position items from left to right and top to bottom according to how important or frequently used they are?

I agree it's important to have people create an account and log in, but once they're logged in they'll never need to click on these items any more, so I'd vote for moving them to the bottom.

gitduks, 2015-12-02:
The idea was to create new groups that clearly belong together. I am not sure what is better in this case. We could also try to combine the 'login' and 'create an account' into a single dialog box that accommodates both.
Btw, I did not mean to get rid of the project notes, they should be at the bottom.

ShuraChechel, 2016-01-05:
Since we have the new Login as part of the main UI, we can put the login, create an account and start tutorial at the bottom.

[image]

bromagosa, 2016-01-26:
I'm on it! What's happened with "Save and share"? Should I remove it?

commit 5b6779f1 (2016-01-26): fixes #192, but I kept 'Change Password' until I implement the new login menu
closed 2016-01-26 by bromagosa