ericrosenbaum/BeetleBlocks #205: Zebra coloring bug

bromagosa, 2016-01-29:
It seems a bug was introduced sometime in recent commits, as I don't recall this happening before:

[image]

If you take the embedded block out and put it back in, it shows up as it should:

[image]

jmoenig, 2016-01-29:
not sure whether you have the latest round of little bug fixes from Snap. I've recently introduced a bunch of zebra-coloring bugs in optimizations which you might have pulled, and then I've fixed them again :-)
Sorry!

bromagosa, 2016-01-29:
Oh I see, I'm pulling the latest changes then. Thanks, this was an easy one! :)

bromagosa, 2016-01-29:
Hmm I did pull the latest changes and it's still there. I guess I proxied the function that fixes it... I'll take a look later!

commit 43f7e7bb (2016-02-01): fixes #205
closed 2016-02-01 by bromagosa