ericrosenbaum/BeetleBlocks #207: new primitives

ShuraChechel, 2016-03-03:
[image]

bromagosa, 2016-03-04:
How to make pyramids in Three.js: http://stackoverflow.com/a/26436728/599939

Some examples including a cone, a truncated cone and a pyramid: https://stemkoski.github.io/Three.js/Shapes.html

In particular, we do have ```THREE.TetrahedronGeometry```, and applying scale transformations to it would give us any kind of pyramid.

As for cones, they're just particular cases of cylinders: ```new THREE.CylinderGeometry( 0, 30, 100, 20, 4 )```, and truncating them is also a particular case:
```
    // radiusAtTop, radiusAtBottom, height, segmentsAroundRadius, segmentsAlongHeight,
    new THREE.CylinderGeometry( 10, 30, 100, 20, 4 )
```

Wedges are different animals, I believe the easiest way would be to build them out of simple triangle extrusions.

bromagosa, 2016-03-04:
Ok, and here's the wedge:

```
Process.prototype.wedge = function(length, width, height, type) {

    // Change positions according to width, height and type
    var triangleShape = new THREE.Shape();
				shape.moveTo(8, 2);
				shape.lineTo(4, 8);
				shape.lineTo(12, 8);
				shape.lineTo(8, 2);

   // The rest is already correct:
    var wedgeGeometry = new THREE.ExtrudeGeometry( triangleShape, { amount: length, steps: 1} );

    var mesh = new THREE.Mesh(wedgeGeometry, beetle.makeMaterial());
					mesh.position.copy(beetle.position);
					mesh.rotation.copy(beetle.rotation);
   
    // Missing the usual reRender() etc.
    [ ... ]
};
```

bromagosa, 2016-06-09:
A wedge implementation: http://beetleblocks.com/run/#present:Username=bromagosa&ProjectName=wedge

gitduks, 2016-06-13:
Interesting general case. I would have added a let/right menu pulldown. It also does not comply with the centroid we usually have...

bromagosa, 2019-11-09:
Proposal: have these as an "extra shapes" library.