ericrosenbaum/BeetleBlocks #212: My Blocks and Variable disappear after saving and returning.

killercatfish, 2016-06-01:
My students keep getting an obsolete block in place of their custom blocks they are creating when they return the next day!  Also, they need to re-create variables when they return--but the instances remain in the code.

ericrosenbaum, 2016-06-01:
thanks for your bug report! sorry to hear you're having trouble, and thanks for your persistence. we'll have a look. @bromagosa any ideas? have you seen these issues?

bromagosa, 2016-06-01:
Hi @killercatfish, I suspect where this issue is coming from... I'm sorry!

Let me try to apply a quick hotfix and get back to you...

closed 2016-06-01 by bromagosa

bromagosa, 2016-06-01:
Yep, that was my fault. It should work now. Please clear your browser cache and reload Beetle Blocks.

Unfortunately, projects that you saved before the fix will not load properly... I'm really sorry about that, it was a side effect I didn't anticipate of something bigger I'm working on :(

commit b450cce7 (2016-06-01): hopefully fixes #212