ericrosenbaum/BeetleBlocks #217: Can't download .stl Files.

Jujulian100, 2016-09-29:
I am on OS X on Chrome. When ever I want to download the .stl File to 3D print, the screen freezes for a while and nothing happens next. When I download .stl (binary) Files it downloads it but on the file it says "Failed - No File". 
Please fix this problem it would be great. 
Best reards
Jujulian100

bromagosa, 2016-09-30:
Hi, would you mind sharing a link to your project so I can take a look? Or the XML file, if you prefer.

Jujulian100, 2016-09-30:
Of course copy past it. It does not support the file it's self. 

<project name="bracelet_Slim" app="BeetleBlocks 1.0, http://beetleblocks.com" version="1"><notes></notes><thumbnail>data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAKAAAAB4CAYAAAB1ovlvAAACJUlEQVR4Xu3SwQkAIAADMd1/6AoucZ90gcKRu23HFIgKXACj8m5/AQBBSAsAmOZ3DiADaQEA0/zOAWQgLQBgmt85gAykBQBM8zsHkIG0AIBpfucAMpAWADDN7xxABtICAKb5nQPIQFoAwDS/cwAZSAsAmOZ3DiADaQEA0/zOAWQgLQBgmt85gAykBQBM8zsHkIG0AIBpfucAMpAWADDN7xxABtICAKb5nQPIQFoAwDS/cwAZSAsAmOZ3DiADaQEA0/zOAWQgLQBgmt85gAykBQBM8zsHkIG0AIBpfucAMpAWADDN7xxABtICAKb5nQPIQFoAwDS/cwAZSAsAmOZ3DiADaQEA0/zOAWQgLQBgmt85gAykBQBM8zsHkIG0AIBpfucAMpAWADDN7xxABtICAKb5nQPIQFoAwDS/cwAZSAsAmOZ3DiADaQEA0/zOAWQgLQBgmt85gAykBQBM8zsHkIG0AIBpfucAMpAWADDN7xxABtICAKb5nQPIQFoAwDS/cwAZSAsAmOZ3DiADaQEA0/zOAWQgLQBgmt85gAykBQBM8zsHkIG0AIBpfucAMpAWADDN7xxABtICAKb5nQPIQFoAwDS/cwAZSAsAmOZ3DiADaQEA0/zOAWQgLQBgmt85gAykBQBM8zsHkIG0AIBpfucAMpAWADDN7xxABtICAKb5nQPIQFoAwDS/cwAZSAsAmOZ3DiADaQEA0/zOAWQgLQBgmt/5A4Lj3qhsHKpPAAAAAElFTkSuQmCC</thumbnail><stage name="Stage" width="720" height="540" costume="0" tempo="60" threadsafe="false" lines="round" codify="false" inheritance="false" sublistIDs="false" scheduled="false" id="1"><pentrails>data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAtAAAAIcCAYAAADffZlTAAAfFklEQVR4Xu3WoREAAAjEMNh/aVagPuhXOUR3HAECBAgQIECAAAECb4F9Lw0JECBAgAABAgQIEBgB7QkIECBAgAABAgQIBAEBHbBMCRAgQIAAAQIECAhoP0CAAAECBAgQIEAgCAjogGVKgAABAgQIECBAQED7AQIECBAgQIAAAQJBQEAHLFMCBAgQIECAAAECAtoPECBAgAABAgQIEAgCAjpgmRIgQIAAAQIECBAQ0H6AAAECBAgQIECAQBAQ0AHLlAABAgQIECBAgICA9gMECBAgQIAAAQIEgoCADlimBAgQIECAAAECBAS0HyBAgAABAgQIECAQBAR0wDIlQIAAAQIECBAgIKD9AAECBAgQIECAAIEgIKADlikBAgQIECBAgAABAe0HCBAgQIAAAQIECAQBAR2wTAkQIECAAAECBAgIaD9AgAABAgQIECBAIAgI6IBlSoAAAQIECBAgQEBA+wECBAgQIECAAAECQUBAByxTAgQIECBAgAABAgLaDxAgQIAAAQIECBAIAgI6YJkSIECAAAECBAgQENB+gAABAgQIECBAgEAQENABy5QAAQIECBAgQICAgPYDBAgQIECAAAECBIKAgA5YpgQIECBAgAABAgQEtB8gQIAAAQIECBAgEAQEdMAyJUCAAAECBAgQICCg/QABAgQIECBAgACBICCgA5YpAQIECBAgQIAAAQHtBwgQIECAAAECBAgEAQEdsEwJECBAgAABAgQICGg/QIAAAQIECBAgQCAICOiAZUqAAAECBAgQIEBAQPsBAgQIECBAgAABAkFAQAcsUwIECBAgQIAAAQIC2g8QIECAAAECBAgQCAICOmCZEiBAgAABAgQIEBDQfoAAAQIECBAgQIBAEBDQAcuUAAECBAgQIECAgID2AwQIECBAgAABAgSCgIAOWKYECBAgQIAAAQIEBLQfIECAAAECBAgQIBAEBHTAMiVAgAABAgQIECAgoP0AAQIECBAgQIAAgSAgoAOWKQECBAgQIECAAAEB7QcIECBAgAABAgQIBAEBHbBMCRAgQIAAAQIECAhoP0CAAAECBAgQIEAgCAjogGVKgAABAgQIECBAQED7AQIECBAgQIAAAQJBQEAHLFMCBAgQIECAAAECAtoPECBAgAABAgQIEAgCAjpgmRIgQIAAAQIECBAQ0H6AAAECBAgQIECAQBAQ0AHLlAABAgQIECBAgICA9gMECBAgQIAAAQIEgoCADlimBAgQIECAAAECBAS0HyBAgAABAgQIECAQBAR0wDIlQIAAAQIECBAgIKD9AAECBAgQIECAAIEgIKADlikBAgQIECBAgAABAe0HCBAgQIAAAQIECAQBAR2wTAkQIECAAAECBAgIaD9AgAABAgQIECBAIAgI6IBlSoAAAQIECBAgQEBA+wECBAgQIECAAAECQUBAByxTAgQIECBAgAABAgLaDxAgQIAAAQIECBAIAgI6YJkSIECAAAECBAgQENB+gAABAgQIECBAgEAQENABy5QAAQIECBAgQICAgPYDBAgQIECAAAECBIKAgA5YpgQIECBAgAABAgQEtB8gQIAAAQIECBAgEAQEdMAyJUCAAAECBAgQICCg/QABAgQIECBAgACBICCgA5YpAQIECBAgQIAAAQHtBwgQIECAAAECBAgEAQEdsEwJECBAgAABAgQICGg/QIAAAQIECBAgQCAICOiAZUqAAAECBAgQIEBAQPsBAgQIECBAgAABAkFAQAcsUwIECBAgQIAAAQIC2g8QIECAAAECBAgQCAICOmCZEiBAgAABAgQIEBDQfoAAAQIECBAgQIBAEBDQAcuUAAECBAgQIECAgID2AwQIECBAgAABAgSCgIAOWKYECBAgQIAAAQIEBLQfIECAAAECBAgQIBAEBHTAMiVAgAABAgQIECAgoP0AAQIECBAgQIAAgSAgoAOWKQECBAgQIECAAAEB7QcIECBAgAABAgQIBAEBHbBMCRAgQIAAAQIECAhoP0CAAAECBAgQIEAgCAjogGVKgAABAgQIECBAQED7AQIECBAgQIAAAQJBQEAHLFMCBAgQIECAAAECAtoPECBAgAABAgQIEAgCAjpgmRIgQIAAAQIECBAQ0H6AAAECBAgQIECAQBAQ0AHLlAABAgQIECBAgICA9gMECBAgQIAAAQIEgoCADlimBAgQIECAAAECBAS0HyBAgAABAgQIECAQBAR0wDIlQIAAAQIECBAgIKD9AAECBAgQIECAAIEgIKADlikBAgQIECBAgAABAe0HCBAgQIAAAQIECAQBAR2wTAkQIECAAAECBAgIaD9AgAABAgQIECBAIAgI6IBlSoAAAQIECBAgQEBA+wECBAgQIECAAAECQUBAByxTAgQIECBAgAABAgLaDxAgQIAAAQIECBAIAgI6YJkSIECAAAECBAgQENB+gAABAgQIECBAgEAQENABy5QAAQIECBAgQICAgPYDBAgQIECAAAECBIKAgA5YpgQIECBAgAABAgQEtB8gQIAAAQIECBAgEAQEdMAyJUCAAAECBAgQICCg/QABAgQIECBAgACBICCgA5YpAQIECBAgQIAAAQHtBwgQIECAAAECBAgEAQEdsEwJECBAgAABAgQICGg/QIAAAQIECBAgQCAICOiAZUqAAAECBAgQIEBAQPsBAgQIECBAgAABAkFAQAcsUwIECBAgQIAAAQIC2g8QIECAAAECBAgQCAICOmCZEiBAgAABAgQIEBDQfoAAAQIECBAgQIBAEBDQAcuUAAECBAgQIECAgID2AwQIECBAgAABAgSCgIAOWKYECBAgQIAAAQIEBLQfIECAAAECBAgQIBAEBHTAMiVAgAABAgQIECAgoP0AAQIECBAgQIAAgSAgoAOWKQECBAgQIECAAAEB7QcIECBAgAABAgQIBAEBHbBMCRAgQIAAAQIECAhoP0CAAAECBAgQIEAgCAjogGVKgAABAgQIECBAQED7AQIECBAgQIAAAQJBQEAHLFMCBAgQIECAAAECAtoPECBAgAABAgQIEAgCAjpgmRIgQIAAAQIECBAQ0H6AAAECBAgQIECAQBAQ0AHLlAABAgQIECBAgICA9gMECBAgQIAAAQIEgoCADlimBAgQIECAAAECBAS0HyBAgAABAgQIECAQBAR0wDIlQIAAAQIECBAgIKD9AAECBAgQIECAAIEgIKADlikBAgQIECBAgAABAe0HCBAgQIAAAQIECAQBAR2wTAkQIECAAAECBAgIaD9AgAABAgQIECBAIAgI6IBlSoAAAQIECBAgQEBA+wECBAgQIECAAAECQUBAByxTAgQIECBAgAABAgLaDxAgQIAAAQIECBAIAgI6YJkSIECAAAECBAgQENB+gAABAgQIECBAgEAQENABy5QAAQIECBAgQICAgPYDBAgQIECAAAECBIKAgA5YpgQIECBAgAABAgQEtB8gQIAAAQIECBAgEAQEdMAyJUCAAAECBAgQICCg/QABAgQIECBAgACBICCgA5YpAQIECBAgQIAAAQHtBwgQIECAAAECBAgEAQEdsEwJECBAgAABAgQICGg/QIAAAQIECBAgQCAICOiAZUqAAAECBAgQIEBAQPsBAgQIECBAgAABAkFAQAcsUwIECBAgQIAAAQIC2g8QIECAAAECBAgQCAICOmCZEiBAgAABAgQIEBDQfoAAAQIECBAgQIBAEBDQAcuUAAECBAgQIECAgID2AwQIECBAgAABAgSCgIAOWKYECBAgQIAAAQIEBLQfIECAAAECBAgQIBAEBHTAMiVAgAABAgQIECAgoP0AAQIECBAgQIAAgSAgoAOWKQECBAgQIECAAAEB7QcIECBAgAABAgQIBAEBHbBMCRAgQIAAAQIECAhoP0CAAAECBAgQIEAgCAjogGVKgAABAgQIECBAQED7AQIECBAgQIAAAQJBQEAHLFMCBAgQIECAAAECAtoPECBAgAABAgQIEAgCAjpgmRIgQIAAAQIECBAQ0H6AAAECBAgQIECAQBAQ0AHLlAABAgQIECBAgICA9gMECBAgQIAAAQIEgoCADlimBAgQIECAAAECBAS0HyBAgAABAgQIECAQBAR0wDIlQIAAAQIECBAgIKD9AAECBAgQIECAAIEgIKADlikBAgQIECBAgAABAe0HCBAgQIAAAQIECAQBAR2wTAkQIECAAAECBAgIaD9AgAABAgQIECBAIAgI6IBlSoA [... cut ...] <l>1.5</l></block><block s="doRepeat"><l>1</l><script><block s="doRepeat"><l>144</l><script><block s="move"><l>0.325</l></block><block s="rotate"><l><option>y</option></l><l>5.25</l></block><block s="sphere"><l>0.4</l></block></script></block><block s="rotate"><l><option>y</option></l><l>5</l></block><block s="move"><l>0.3</l></block></script></block></script></scripts></sprite></sprites></stage><hidden></hidden><headers></headers><code></code><blocks></blocks><variables></variables></project><<

bromagosa, 2016-09-30:
This is just an image encoded in base64, not the project contents...

If GitHub doesn't let you attach XMLs, you can just make the project public (file, open, select the project, click on share) and send me the URL.

Jujulian100, 2016-09-30:
[http://beetleblocks.com/run/#present:Username=jujulian100&ProjectName=bracelet_Slim](url)

Jujulian100, 2016-09-30:
[http://beetleblocks.com/run/#present:Username=Jujulian100&ProjectName=bracelet_Slim](url)

Jujulian100, 2016-09-30:
this is the URL but it says error 404

bromagosa, 2016-09-30:
The reason you can't download this project is that there are too many objects in the scene, namely 46008 spheres.

In the future we may switch to a different way of representing geometry that doesn't treat each shape as a separate object, but for now you should probably try to modify your code so that it spawns less spheres...

Sorry about that :(

Jujulian100, 2016-09-30:
Thank you for telling me that. 
Really love the concept though.

ericrosenbaum, 2016-09-30:
Hi @Jujulian100 - thanks for the bug report, and sorry to hear you're having trouble. I like the bracelet design! I agree with @bromagosa that it cannot create the STL file because there are too many spheres. I remixed your project, and removed a part that generates about 40,000 spheres, but it still looks about the same:

[http://beetleblocks.com/run/#present:Username=ericrosenbizzle&ProjectName=bracelet_Slim%20remix](http://beetleblocks.com/run/#present:Username=ericrosenbizzle&ProjectName=bracelet_Slim%20remix)

Interestingly, the regular STL download still doesn't download, but the binary STL does. the binary STL is still really big (97MB) and seems to crash meshlab for me... I tried replacing the spheres with cubes though and that worked.

Jujulian100, 2016-09-30:
Thank you so much. 
 .D