ericrosenbaum/BeetleBlocks #220: Square extrusion

mrspollard, 2016-12-04:
Hi Bernat,

I was wondering if you could add a square shape (flat bottom) extrusion choice?  It would help with 3D printing and eliminate the need to make multiple thousands of really small, cuboids.

The round tube extrusions aren't ideal for 3d printing and larger file size of so many small cuboids are difficult for 3D slicers.

Thanks.

bromagosa, 2016-12-05:
Hi @mrspollard!

That's something that's been on our minds for a while, but we want to get it right and make it abstract enough so that any 2D shape can be extruded.

At the moment my limited time is being spent on getting the beta site ready, but we'll tackle this at our next meeting and see if we can come up with something.

mrspollard, 2016-12-05:
Thanks Bernat,

Is this something you'd rather I post on the beta site forum, or is here okay?

Laurel

bromagosa, 2016-12-07:
I think it doesn't matter much until we officially launch the beta site, but feel free to post questions/suggestions there if you want :)