ericrosenbaum/BeetleBlocks #231: Won't Save

bwmartin44, 2018-07-25:
I have a group of students using a shared account.  They are trying to save their projects, but when they save to the cloud, none of them are showing up on the My Projects page.

bwmartin44, 2018-07-25:
On a different account I can access old saved files but nothing new will save.

bromagosa, 2018-07-26:
Sorry, I was travelling. Is this still an issue?

If so, can you let me know the account name? I'll see what I can find out.

bwmartin44, 2018-07-26:
vmsfablab

thanks for looking into it

bromagosa, 2018-08-06:
Hi @bwmartin44, and sorry about the delay. I've been in a conference and then on a short vacation.

I've now fixed the issue. Thank you very much for reporting it and please excuse the trouble!

closed 2018-08-06 by bromagosa