ericrosenbaum/BeetleBlocks #232: Project from 2017 now has 'obsolete' blocks

ToonTalk, 2018-08-01:
http://beetleblocks.com/users/toontalk/projects/program%20interpolator

no longer works. 

[image]

While it used to look like this

[image]

A clue to what is going is that I can find the missing blocks

[image]

But note that these are gray and I probably imported them from Snap!. These "missing" blocks I believe are still in the XML file.

The project stopped working some time this year.

bromagosa, 2019-11-09:
Hi, Ken.

Can you please share the xml with us? I think I can make the project work for you...

ToonTalk, 2019-11-12:
Hi. Thanks.

I needed to demonstrate this at Constructionism 2018 and found out that I had an old copy of BeetleBlocks that ran it fine - https://ecraft2learn.github.io/uui/beetleblocks/

The attached works fine in that version but has the same problem when imported from http://beetleblocks.com/run/# as well as a new problem:

Uncaught TypeError: Cannot read property 'drawImage' of null
    at Image.project.pentrails.onload (store.js:399)
[constructionism 2.zip]

(https://github.com/ericrosenbaum/BeetleBlocks/files/3836250/constructionism.2.zip)

bromagosa, 2019-11-13:
Hi Ken,

the issue comes from the blocks in the "other" category, which doesn't exist in Beetle Blocks. I've manually moved them to "My blocks" and it now works okay.

Here's the fixed version: [constructionism 2.zip](https://github.com/ericrosenbaum/BeetleBlocks/files/3841172/constructionism.2.zip)

closed 2019-11-13 by bromagosa

ToonTalk, 2019-11-13:
Thanks - unshared the broken version and shared the working one.