ericrosenbaum/BeetleBlocks #234: Letters not showing when running offline

sfederici, 2018-08-10:
I'm using beetleblocks offline in a school with no internet connection. I found out that the names of the axis show up as black squares and that text created by the "text" block don't show up at all. The same project, when loaded in the beetleblocks website, works perfectly well. Is there something I can do or that I can add to the beetleblocks folder in order to make the project work well even offline?
Thanks in advance
[spazio_tridimensionale.zip](https://github.com/ericrosenbaum/BeetleBlocks/files/2278390/spazio_tridimensionale.zip)

bromagosa, 2019-11-09:
Hi Stefano.

Have you tried this recently? It works fine for me...

sfederici, 2019-11-10:
Hi Bernat. I had solved the problem by using a very simple local web server, https://www.ritlabs.com/en/products/tinyweb/. It is very small and works very well with beetleblocks.

closed 2019-11-10 by sfederici